ldrobotSensorTeam/ldlidar_stl_ros2 #1: LD06 - Stop the sensor motor

alcor55, 2022-04-05:
Hi, do you know if there is any kind of way to set the lidar in a 'stand by' mode? In order to reduce the lidar's  power consumption and mechanical parts by stoping the laser head rotation. The goal it to enable it only when it is in use.

ldrobotsensor, 2022-04-06:
Hello, at present, the best way for our products to meet the functions of stopping and reducing power consumption is to disconnect the power

ldrobotsensor, 2022-04-06:
@alcor55 Reply see above

alcor55, 2022-04-06:
Thank you for answering. But, in the developer's manual it is written that the dc motor can stop the rotation. Is there any software command to shutdown? I'm worry, would the sensor get damaged by disconection of the power? (Hot swap)

ldrobotsensor, 2022-04-06:
Hello, our products can stop the motor, but the radar cannot be powered off when the power supply is continuous. For the motor to stop, input a PWM signal to it to make it work in the external speed control mode, and input a smaller value of PWM The duty cycle value, but it means that the radar speed closed-loop control needs to be carried out externally. The specific implementation is described in the development manual. The following link can be downloaded to the latest development manual.
https://github.com/ldrobotSensorTeam/ldlidar_stl_ros/releases/tag/v2.2.5
@alcor55

alcor55, 2022-04-06:
Thanks a lot, you can close the issue.

alcor55, 2022-04-06:
Sorry, une last question: is not possible to input a PWM signal to it to make it work in the external speed control mode, there re-enable the internal PWM with another signal?

ldrobotsensor, 2022-04-06:
Hello, when entering the external speed control mode, the internal speed control mode will only be entered when the lidar is powered off and there is no PWM signal input after power-on. @alcor55

ldrobotsensor, 2022-04-06:
## See the development manual for details：
---
[image]

alcor55, 2022-04-06:
Thanks anyway I will search for another solution.

ldrobotsensor, 2022-04-06:
OK, any comments are welcome