ldrobotSensorTeam/ldlidar_stl_ros2 #2: Error when launching python code.

Kristian181, 2022-11-17:
After the command "ros2 launch ldlidar_stl_ros2 ld19.launch.py" I recieved the following error:

[INFO] [launch]: All log files can be found below /home/ditmar/.ros/log/2022-11-17-10-38-42-823400-ditmar-VirtualBox-3613 
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [ldlidar_stl_ros2_node-1]: process started with pid [3614]
[INFO] [static_transform_publisher-2]: process started with pid [3616]
[static_transform_publisher-2] [WARN] [1668677924.187515024] []: Old-style arguments are deprecated; see --help for new-style arguments
[ldlidar_stl_ros2_node-1] [INFO] [1668677924.433554313] [LD19]: LDLiDAR SDK Pack Version is: v3.0.3
[ldlidar_stl_ros2_node-1] [INFO] [1668677924.434860673] [LD19]: <product_name>: LDLiDAR_LD19
[ldlidar_stl_ros2_node-1] [INFO] [1668677924.435963707] [LD19]: <topic_name>: scan
[ldlidar_stl_ros2_node-1] [INFO] [1668677924.436106027] [LD19]: <frame_id>: base_laser
[ldlidar_stl_ros2_node-1] [INFO] [1668677924.436180008] [LD19]: <port_name>: /dev/ttyUSB0
[ldlidar_stl_ros2_node-1] [INFO] [1668677924.436199364] [LD19]: <port_baudrate>: 230400
[ldlidar_stl_ros2_node-1] [INFO] [1668677924.436209824] [LD19]: <laser_scan_dir>: Counterclockwise
[ldlidar_stl_ros2_node-1] [INFO] [1668677924.436219813] [LD19]: <enable_angle_crop_func>: false
[ldlidar_stl_ros2_node-1] [INFO] [1668677924.436229702] [LD19]: <angle_crop_min>: 135.000000
[ldlidar_stl_ros2_node-1] [INFO] [1668677924.436245783] [LD19]: <angle_crop_max>: 225.000000
[ldlidar_stl_ros2_node-1] [ERROR] [1668677924.436360430] [LD19]: ldlidar node start is fail
[static_transform_publisher-2] [INFO] [1668677924.503143437] [base_link_to_base_laser_ld19]: Spinning until stopped - publishing transform
[static_transform_publisher-2] translation: ('0.000000', '0.000000', '0.180000')
[static_transform_publisher-2] rotation: ('0.000000', '0.000000', '0.000000', '1.000000')
[static_transform_publisher-2] from 'base_link' to 'base_laser'
[ldlidar_stl_ros2_node-1] [LDS][ERROR][Thu Nov 17 10:38:44 2022][/home/ditmar/ldlidar_ros2_ws/src/ldlidar_stl_ros2/ldlidar_driver/src/serialcom/serial_interface_linux.cpp][Open][41][Open open error,Permission denied]
[ldlidar_stl_ros2_node-1] [LDS][ERROR][Thu Nov 17 10:38:44 2022][/home/ditmar/ldlidar_ros2_ws/src/ldlidar_stl_ros2/ldlidar_driver/src/core/ldlidar_driver.cpp][Start][90][serial is not open:/dev/ttyUSB0]
[ERROR] [ldlidar_stl_ros2_node-1]: process has died [pid 3614, exit code 1, cmd '/home/ditmar/ldlidar_ros2_ws/install/ldlidar_stl_ros2/lib/ldlidar_stl_ros2/ldlidar_stl_ros2_node --ros-args -r __node:=LD19 --params-file /tmp/launch_params_02h4cjfi --params-file /tmp/launch_params_4p_xhzqx --params-file /tmp/launch_params_crh_pjiy --params-file /tmp/launch_params_d3w9fnwj --params-file /tmp/launch_params_ed2i03x1 --params-file /tmp/launch_params_xthq6i0l --params-file /tmp/launch_params_kcwvvbzd --params-file /tmp/launch_params_3_0jvih8 --params-file /tmp/launch_params_taoamxlv'].

Currently using ubuntu version: Ubuntu 22.04.1 LTS
I'm also using Oracle Virtual Machine to run Ubuntu.

I hope somone can help me with this problem.

ldrobotsensor, 2022-11-17:
Set the -x permission for the serial port device mounted by the lidar in the system:
`sudo chmod 777 /dev/ttyUSB*`

Kristian181, 2022-11-17:
Thanks for the quick response!

The sudo chmod 777 /dev/ttyUSB* seems to have fixed the permission problem.
But after the following text it stops printing. 

ditmar@ditmar-VirtualBox:~/ldlidar_ros2_ws$ ros2 launch ldlidar_stl_ros2 ld19.launch.py 
[INFO] [launch]: All log files can be found below /home/ditmar/.ros/log/2022-11-17-12-21-39-380464-ditmar-VirtualBox-5986
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [ldlidar_stl_ros2_node-1]: process started with pid [5987]
[INFO] [static_transform_publisher-2]: process started with pid [5989]
[static_transform_publisher-2] [WARN] [1668684099.539221234] []: Old-style arguments are deprecated; see --help for new-style arguments
[static_transform_publisher-2] [INFO] [1668684099.581556904] [base_link_to_base_laser_ld19]: Spinning until stopped - publishing transform
[static_transform_publisher-2] translation: ('0.000000', '0.000000', '0.180000')
[static_transform_publisher-2] rotation: ('0.000000', '0.000000', '0.000000', '1.000000')
[static_transform_publisher-2] from 'base_link' to 'base_laser'
[ldlidar_stl_ros2_node-1] [INFO] [1668684099.611595903] [LD19]: LDLiDAR SDK Pack Version is: v3.0.3
[ldlidar_stl_ros2_node-1] [INFO] [1668684099.612532324] [LD19]: <product_name>: LDLiDAR_LD19
[ldlidar_stl_ros2_node-1] [INFO] [1668684099.613199108] [LD19]: <topic_name>: scan
[ldlidar_stl_ros2_node-1] [INFO] [1668684099.613551183] [LD19]: <frame_id>: base_laser
[ldlidar_stl_ros2_node-1] [INFO] [1668684099.613906238] [LD19]: <port_name>: /dev/ttyUSB0
[ldlidar_stl_ros2_node-1] [INFO] [1668684099.614215534] [LD19]: <port_baudrate>: 230400
[ldlidar_stl_ros2_node-1] [INFO] [1668684099.614502757] [LD19]: <laser_scan_dir>: Counterclockwise
[ldlidar_stl_ros2_node-1] [INFO] [1668684099.614756829] [LD19]: <enable_angle_crop_func>: false
[ldlidar_stl_ros2_node-1] [INFO] [1668684099.615068479] [LD19]: <angle_crop_min>: 135.000000
[ldlidar_stl_ros2_node-1] [INFO] [1668684099.615259426] [LD19]: <angle_crop_max>: 225.000000
[ldlidar_stl_ros2_node-1] [INFO] [1668684100.083317173] [LD19]: ldlidar node start is success
[ldlidar_stl_ros2_node-1] [INFO] [1668684100.110435328] [LD19]: ldlidar communication is normal.
[ldlidar_stl_ros2_node-1] [INFO] [1668684100.116253038] [LD19]: Publish topic message:ldlidar scan data.

Also the /scan doesn't show up in the ros2 topic list.

ditmar@ditmar-VirtualBox:~$ ros2 topic list 
/parameter_events
/rosout

Kristian181, 2022-11-17:
It works now, after a second time the /scan did show up.
I am now able to vizualize the data from the lidar in rviz2. 

I want to use the lidar to detect obstacles on a modol car.
Is ther a way to print/obtain the "raw data" like Distance, Start angle and End angle?
So I can use the data in code.