ldrobotSensorTeam/ldlidar_stl_ros2 #4: LaserRangeScan readings

yorobot256, 2023-01-10:
Hi,

I am trying to do a mapping with slam_toolbox and LD06 lidar, but I having this error :

[async_slam_toolbox_node-1] LaserRangeScan contains 455 range readings, expected 460
[async_slam_toolbox_node-1] LaserRangeScan contains 455 range readings, expected 460
[async_slam_toolbox_node-1] LaserRangeScan contains 455 range readings, expected 460
[async_slam_toolbox_node-1] LaserRangeScan contains 455 range readings, expected 460
[async_slam_toolbox_node-1] LaserRangeScan contains 454 range readings, expected 460
[async_slam_toolbox_node-1] LaserRangeScan contains 454 range readings, expected 460
[async_slam_toolbox_node-1] LaserRangeScan contains 455 range readings, expected 460
[async_slam_toolbox_node-1] LaserRangeScan contains 455 range readings, expected 460
[async_slam_toolbox_node-1] LaserRangeScan contains 454 range readings, expected 460
[async_slam_toolbox_node-1] LaserRangeScan contains 453 range readings, expected 460

Slam_Toolbox expected a fixed number of points from Lidar, but for some reason LD06 is not providing.
Do you know how this be fixed on LD06 Ros2 package?

I similar issue I found, but for a different lidar : https://github.com/SteveMacenski/slam_toolbox/issues/278

Thanks
Yoan

AricTDS, 2023-01-11:
Hi.
Thank you very much for your question

I have read your message carefully. Judging from your error, the problem you reported is that there was an error when you imported sensor data in the Slam_Toolbox.

[async_slam_toolbox_node-1] LaserRangeScan contains 455 range readings, expected 460

Since I have never used slam_toolbox tool for verification, through your link [,](SteveMacenski/slam_toolbox#278) I see other friends have the same problem, but they all have different expectations. What I understand about this expected number is, The Angle increment parameter passed in by the subject driven by the sensor is calculated as 460. I'm not sure about this one. If you are clear, please help to explain the fixed number of Slam_Toolbox 460, which data is calculated through.

The Lidar data is not fixed, because the driver may have some filtering algorithm or speed problems, the data amount of each frame may be different, I would like to confirm here.

AricTDS, 2023-01-11:
Hi
I confirm that due to the sensor design, the motor speed cannot be 100% consistent, so the total number of each circle may be different. 
Regarding your use of slam_toolbox tool, the expected number of 460 is inconsistent, I suggest that you can make up 460 points by manually filling zeros.

yorobot256, 2023-01-11:
Hi AricTDS,

Thanks for your reply, I will try your suggested solution.

While I was reviewing the code, I saw this condition in a file:
// wait for enough data, need enough data to show a circle
    // enough data has been obtained
    if ((n.angle < 20.0) && (last_angle > 340.0)) {
I am wondering if we can play with last_angle parameter, like increasing to 350 or 355, it will allow to get more complete readings ?

Thanks and Regards,

AricTDS, 2023-01-12:
Here is the threshold judgment based on the characteristics of the sensor to ensure that the original data analyzed are valid data.

At the same time, for the characteristics of this sensor, even if there are only about 450 points in a circle, even if the Angle threshold is adjusted, it is not necessarily guaranteed to be 460 fixed points.

But we can guarantee that the final data are valid data, so if the number is less than 460, it is good to manually fill the zero.

sskorol, 2023-02-05:
Seeing the same, but `LaserRangeScan contains 454 range readings, expected 455`. Note that 454 might be in a range between 453 and 456, according to my observations.
@AricTDS could you please point out the piece of code that should be updated to do this trick?

AricTDS, 2023-02-07:
> Seeing the same, but `LaserRangeScan contains 454 range readings, expected 455`. Note that 454 might be in a range between 453 and 456, according to my observations. @AricTDS could you please point out the piece of code that should be updated to do this trick with zeros?
----------------------------------------------------------------------------

I suggest that you can check whether the size of the cloud packet meets the requirements before publishing the point cloud data, and fill in the zero if it does not meet the requirements. It's okay if you have a better way.

Chambana, 2023-07-04:
I'd also like to see an option for consistent 460 point readings, with the difference filled by 0's.

DrewSBAI, 2023-08-19:
@sskorol @yorobot256 @AricTDS  also see https://github.com/ldrobotSensorTeam/ldlidar_stl_ros2/issues/11

yorobot256, 2023-09-30:
Some of you may be interested in a solution for this error.

I updated thanks to chatgpt this function to allways return a fixed number of points from LD06 lidar

bool LiPkg::GetLaserScanData(Points2D& out) {
  if (IsFrameReady()) {
    ResetFrameReady();
    {
      std::lock_guard<std::mutex> lg(mutex_lock2_);
      out = laser_scan_data_;
    }

    // Truncate or fill with NaN values
    if (out.size() < 456) {
      const double nan_value = std::numeric_limits<double>::quiet_NaN();
      size_t current_size = out.size();
      out.resize(456, PointData(nan_value, nan_value, nan_value, 0));
      // Set timestamps for the NaN values to 0
      for (size_t i = current_size; i < 456; ++i) {
        out[i].stamp = 0;
      }
    } else if (out.size() > 456) {
      out.resize(456); // Truncate to 456 points
    }

    return true;
  } else {
    return false;
  }
}

sskorol, 2023-10-01:
@yorobot256 so no drops in slam-toolbox anymore?

yorobot256, 2023-10-01:
@sskorol with this fix not anymore. Now I am able to use slam_toolbox and to generate a map. Next step is to try Nav2

bperseghetti, 2023-11-17:
@yorobot256 @sskorol @Chambana @AricTDS @DrewSBAI You can now set binning as a param and it will upscale or downscale to what you want. Also allows for setting area mask so you can mask out returns for transient events (like an elevated magnetometer cable in FOV) works great in Nav2 and reduces CPU load a ton.
Here is the PR: https://github.com/ldrobotSensorTeam/ldlidar_stl_ros2/pull/14

yorobot256, 2024-01-24:
> @yorobot256 @sskorol @Chambana @AricTDS @DrewSBAI You can now set binning as a param and it will upscale or downscale to what you want. Also allows for setting area mask so you can mask out returns [...]

Works well, Thanks

yorobot256, 2024-01-25:
Fixed with binning option

closed 2024-01-25 by yorobot256

Oyefusi-Samuel, 2025-05-03:
slam_toolbox:
  ros__parameters:
    ...
    expected_scan_count: 350