ldrobotSensorTeam/ldlidar_stl_ros2 #5: Error ouccurred  at ros2 launch ld06.launch.py

Hermanye996, 2023-03-07:
'''bash
[ERROR] [ldlidar_stl_ros2_node-1]: process has died [pid 7791, exit code 1, cmd '/home/mangdang/ldlidar_ros2_ws/install/ldlidar_stl_ros2/lib/ldlidar_stl_ros2/ldlidar_stl_ros2_node --ros-args -r __node:=LD06 --params-file /tmp/launch_params_4jbfaw75 --params-file /tmp/launch_params_gydxgavv --params-file /tmp/launch_params_bybutd2d --params-file /tmp/launch_params_b4orcomy --params-file /tmp/launch_params_1yr3i283 --params-file /tmp/launch_params_m3umg0fo --params-file /tmp/launch_params_lwbet75o --params-file /tmp/launch_params_5ss3a7t9 --params-file /tmp/launch_params_u6ozj208'].
'''

I've finished all steps in README, and tried sudo chmod 777 /dev/ttyUSB*
chmod result:
 @```bash
ls -l /dev/ttyUSB0 
crwxrwxrwx 1 root dialout 188, 0  3  7 13:53 /dev/ttyUSB0
```

mingdonghu, 2023-03-07:
> ```
> [ERROR] [ldlidar_stl_ros2_node-1]: process has died [pid 7791, exit code 1, cmd '/home/mangdang/ldlidar_ros2_ws/install/ldlidar_stl_ros2/lib/ldlidar_stl_ros2/ldlidar_stl_ros2_node --ros-args -r __node:=LD06 --params-file /tmp/launch_params_4jbfaw75 --params-file /tmp/launch_params_gydxgavv --params-file /tmp/launch_params_bybutd2d --params-file /tmp/launch_params_b4orcomy --params-file /tmp/launch_params_1yr3i283 --params-file /tmp/launch_params_m3umg0fo --params-file /tmp/launch_params_lwbet75o --params-file /tmp/launch_params_5ss3a7t9 --params-file /tmp/launch_params_u6ozj208'].
> ```
> 
> I've finished all steps in README, and tried sudo chmod 777 /dev/ttyUSB* chmod result:
> 
> ```
> ls -l /dev/ttyUSB0 
> crwxrwxrwx 1 root dialout 188, 0  3  7 13:53 /dev/ttyUSB0
> ```

你好，可以提供完整一点的日志吗？确认连接Lidar的USB在系统上挂载的设备号是否是"/dev/ttyUSB0"。 @Hermanye996

Hermanye996, 2023-03-07:
I unplugged the lidar then replugged it, and confirmed that the device number mounted on Ubuntu is /dev/ttyUSB0.
`
Operation:
ls -l /dev/tty*

Results:
crw-rw-rw- 1 root     tty       5,  0  3月  7 15:09 /dev/tty
crw--w---- 1 root     tty       4,  0  3月  7 12:47 /dev/tty0
crw--w---- 1 root     tty       4,  1  3月  7 12:47 /dev/tty1
crw--w---- 1 root     tty       4, 10  3月  7 12:47 /dev/tty10
crw--w---- 1 root     tty       4, 11  3月  7 12:47 /dev/tty11
crw--w---- 1 root     tty       4, 12  3月  7 12:47 /dev/tty12
crw--w---- 1 root     tty       4, 13  3月  7 12:47 /dev/tty13
crw--w---- 1 root     tty       4, 14  3月  7 12:47 /dev/tty14
crw--w---- 1 root     tty       4, 15  3月  7 12:47 /dev/tty15
crw--w---- 1 root     tty       4, 16  3月  7 12:47 /dev/tty16
crw--w---- 1 root     tty       4, 17  3月  7 12:47 /dev/tty17
crw--w---- 1 root     tty       4, 18  3月  7 12:47 /dev/tty18
crw--w---- 1 root     tty       4, 19  3月  7 12:47 /dev/tty19
crw--w---- 1 mangdang tty       4,  2  3月  7 12:47 /dev/tty2
crw--w---- 1 root     tty       4, 20  3月  7 12:47 /dev/tty20
crw--w---- 1 root     tty       4, 21  3月  7 12:47 /dev/tty21
crw--w---- 1 root     tty       4, 22  3月  7 12:47 /dev/tty22
crw--w---- 1 root     tty       4, 23  3月  7 12:47 /dev/tty23
crw--w---- 1 root     tty       4, 24  3月  7 12:47 /dev/tty24
crw--w---- 1 root     tty       4, 25  3月  7 12:47 /dev/tty25
crw--w---- 1 root     tty       4, 26  3月  7 12:47 /dev/tty26
crw--w---- 1 root     tty       4, 27  3月  7 12:47 /dev/tty27
crw--w---- 1 root     tty       4, 28  3月  7 12:47 /dev/tty28
crw--w---- 1 root     tty       4, 29  3月  7 12:47 /dev/tty29
crw--w---- 1 root     tty       4,  3  3月  7 12:47 /dev/tty3
crw--w---- 1 root     tty       4, 30  3月  7 12:47 /dev/tty30
crw--w---- 1 root     tty       4, 31  3月  7 12:47 /dev/tty31
crw--w---- 1 root     tty       4, 32  3月  7 12:47 /dev/tty32
crw--w---- 1 root     tty       4, 33  3月  7 12:47 /dev/tty33
crw--w---- 1 root     tty       4, 34  3月  7 12:47 /dev/tty34
crw--w---- 1 root     tty       4, 35  3月  7 12:47 /dev/tty35
crw--w---- 1 root     tty       4, 36  3月  7 12:47 /dev/tty36
crw--w---- 1 root     tty       4, 37  3月  7 12:47 /dev/tty37
crw--w---- 1 root     tty       4, 38  3月  7 12:47 /dev/tty38
crw--w---- 1 root     tty       4, 39  3月  7 12:47 /dev/tty39
crw--w---- 1 root     tty       4,  4  3月  7 12:47 /dev/tty4
crw--w---- 1 root     tty       4, 40  3月  7 12:47 /dev/tty40
crw--w---- 1 root     tty       4, 41  3月  7 12:47 /dev/tty41
crw--w---- 1 root     tty       4, 42  3月  7 12:47 /dev/tty42
crw--w---- 1 root     tty       4, 43  3月  7 12:47 /dev/tty43
crw--w---- 1 root     tty       4, 44  3月  7 12:47 /dev/tty44
crw--w---- 1 root     tty       4, 45  3月  7 12:47 /dev/tty45
crw--w---- 1 root     tty       4, 46  3月  7 12:47 /dev/tty46
crw--w---- 1 root     tty       4, 47  3月  7 12:47 /dev/tty47
crw--w---- 1 root     tty       4, 48  3月  7 12:47 /dev/tty48
crw--w---- 1 root     tty       4, 49  3月  7 12:47 /dev/tty49
crw--w---- 1 root     tty       4,  5  3月  7 12:47 /dev/tty5
crw--w---- 1 root     tty       4, 50  3月  7 12:47 /dev/tty50
crw--w---- 1 root     tty       4, 51  3月  7 12:47 /dev/tty51
crw--w---- 1 root     tty       4, 52  3月  7 12:47 /dev/tty52
crw--w---- 1 root     tty       4, 53  3月  7 12:47 /dev/tty53
crw--w---- 1 root     tty       4, 54  3月  7 12:47 /dev/tty54
crw--w---- 1 root     tty       4, 55  3月  7 12:47 /dev/tty55
crw--w---- 1 root     tty       4, 56  3月  7 12:47 /dev/tty56
crw--w---- 1 root     tty       4, 57  3月  7 12:47 /dev/tty57
crw--w---- 1 root     tty       4, 58  3月  7 12:47 /dev/tty58
crw--w---- 1 root     tty       4, 59  3月  7 12:47 /dev/tty59
crw--w---- 1 root     tty       4,  6  3月  7 12:47 /dev/tty6
crw--w---- 1 root     tty       4, 60  3月  7 12:47 /dev/tty60
crw--w---- 1 root     tty       4, 61  3月  7 12:47 /dev/tty61
crw--w---- 1 root     tty       4, 62  3月  7 12:47 /dev/tty62
crw--w---- 1 root     tty       4, 63  3月  7 12:47 /dev/tty63
crw--w---- 1 root     tty       4,  7  3月  7 12:47 /dev/tty7
crw--w---- 1 root     tty       4,  8  3月  7 12:47 /dev/tty8
[... 21 lines cut ...]
crw-rw---- 1 root     dialout   4, 90  3月  7 12:47 /dev/ttyS26
crw-rw---- 1 root     dialout   4, 91  3月  7 12:47 /dev/ttyS27
crw-rw---- 1 root     dialout   4, 92  3月  7 12:47 /dev/ttyS28
crw-rw---- 1 root     dialout   4, 93  3月  7 12:47 /dev/ttyS29
crw-rw---- 1 root     dialout   4, 67  3月  7 12:47 /dev/ttyS3
crw-rw---- 1 root     dialout   4, 94  3月  7 12:47 /dev/ttyS30
crw-rw---- 1 root     dialout   4, 95  3月  7 12:47 /dev/ttyS31
crw-rw---- 1 root     dialout   4, 68  3月  7 12:47 /dev/ttyS4
crw-rw---- 1 root     dialout   4, 69  3月  7 12:47 /dev/ttyS5
crw-rw---- 1 root     dialout   4, 70  3月  7 12:47 /dev/ttyS6
crw-rw---- 1 root     dialout   4, 71  3月  7 12:47 /dev/ttyS7
crw-rw---- 1 root     dialout   4, 72  3月  7 12:47 /dev/ttyS8
crw-rw---- 1 root     dialout   4, 73  3月  7 12:47 /dev/ttyS9
crw-rw---- 1 root     dialout 188,  0  3月  7 16:20 /dev/ttyUSB0
`
then
`
Operation:
sudo chmod 777 /dev/ttyUSB*
ls -l /dev/ttyUSB*

Result:
crwxrwxrwx 1 root dialout 188, 0  3月  7 16:20 /dev/ttyUSB0
`
@ldrobotsensor

mingdonghu, 2023-03-07:
> I unplugged the lidar then replugged it, and confirmed that the device number mounted on Ubuntu is /dev/ttyUSB0.
>
> ```
> Operation:
> ls -l /dev/tty*
>
> Results:
> crw-rw-rw- 1 root     tty       5,  0  3月 [...]

1.运行`ros2 launch ldlidar_stl_ros2 ld06.launch.py` 后输出的日志，可以提供一下吗？
2.激光雷达的灯是否是旋转的吗？
3.你这边使用USB转串口模块是使用官方提供的吗？如果是官方提供的，你观察一下上面的指示灯是否持续闪烁，如果不持续闪烁或者不亮，我建议你这边使用自己的USB转串口模块，单独接线连接到主机上，或者向店家反馈问题
@Hermanye996

Hermanye996, 2023-03-09:
For q1:
`message;severity;node;stamp;location
"ldlidar communication is abnormal.";"40";"LD06";"1678359932.993650353";"/home/mangdang/ldlidar_ros2_ws/src/ldlidar_stl_ros2/src/demo.cpp:main:103"
"ldlidar node start is success";"20";"LD06";"1678359929.993229826";"/home/mangdang/ldlidar_ros2_ws/src/ldlidar_stl_ros2/src/demo.cpp:main:94"
"<angle_crop_max>: 225.000000";"20";"LD06";"1678359929.992244253";"/home/mangdang/ldlidar_ros2_ws/src/ldlidar_stl_ros2/src/demo.cpp:main:78"
"<angle_crop_min>: 135.000000";"20";"LD06";"1678359929.992235376";"/home/mangdang/ldlidar_ros2_ws/src/ldlidar_stl_ros2/src/demo.cpp:main:77"
"<enable_angle_crop_func>: false";"20";"LD06";"1678359929.992230680";"/home/mangdang/ldlidar_ros2_ws/src/ldlidar_stl_ros2/src/demo.cpp:main:76"
"<laser_scan_dir>: Counterclockwise";"20";"LD06";"1678359929.992225667";"/home/mangdang/ldlidar_ros2_ws/src/ldlidar_stl_ros2/src/demo.cpp:main:75"
"<port_baudrate>: 230400";"20";"LD06";"1678359929.992220810";"/home/mangdang/ldlidar_ros2_ws/src/ldlidar_stl_ros2/src/demo.cpp:main:74"
"<port_name>: /dev/ttyUSB0";"20";"LD06";"1678359929.992215880";"/home/mangdang/ldlidar_ros2_ws/src/ldlidar_stl_ros2/src/demo.cpp:main:73"
"<frame_id>: base_laser";"20";"LD06";"1678359929.992211021";"/home/mangdang/ldlidar_ros2_ws/src/ldlidar_stl_ros2/src/demo.cpp:main:72"
"<topic_name>: scan";"20";"LD06";"1678359929.992205471";"/home/mangdang/ldlidar_ros2_ws/src/ldlidar_stl_ros2/src/demo.cpp:main:71"
"<product_name>: LDLiDAR_LD06";"20";"LD06";"1678359929.992195784";"/home/mangdang/ldlidar_ros2_ws/src/ldlidar_stl_ros2/src/demo.cpp:main:70"
"Spinning until stopped - publishing transform
translation: ('0.000000', '0.000000', '0.180000')
rotation: ('0.000000', '0.000000', '0.000000', '1.000000')
from 'base_link' to 'base_laser'";"20";"base_link_to_base_laser_ld06";"1678359929.994488389";"./src/static_transform_broadcaster_program.cpp:main:397"`

For q2:
lidar rotated when I plug it in USB port, then stopped.

For q3:
The USB to serial board is provided by LD official, the constructor light blinked in 2HZ, when I pressed botton on board, the lidar started to rotate. @mingdonghu

mingdonghu, 2023-03-10:
你好，我这边建议：
1. 找一个USB转串口模块(例如cp2102、ch340等)，单独接线，对于接口图可以去看下雷达的datasheet，接入PC；
2. 打开一个串口调试助手，串口参数设置按照datasheet上面来，观察字节数据中是否有很多`0x54 0x2C`,如果有就说明雷达本身是正常的；
3. 按照单独的USB转串口模块连接雷达，然后去运行ROS2功能包
@Hermanye996

mingdonghu, 2023-03-10:
问题可能出在官方串口转接板上
@Hermanye996