ldrobotSensorTeam/ldlidar_stl_ros2 #10: How to make lidar stop scanning

altineller, 2023-08-04:
Hello,

Is there a command I can send to the lidar to make it stop scanning, and still accept serial commands, and  then start scanning again?

Can there be a hardware solution?

Best Regards,
C.A.

alcor55, 2023-10-20:
My case, LD06 lidar. The sensor doesn't have an RX serial pin, so you can't send serial commands. It has a PWM in (pin 2). In theory, if you send a 3.3v PWM signal by the RPi GPIO18 pin, with a duty cycle of 46-54% and frequency 20-50khz, you can drive the sensor motor (start/stop/custom velocity). Though I'm trying it, the lidar ignores completely the PWM signal. I think I should add a mosfet in order to cut off the 5v Vin.

altineller, 2023-10-28:
The Lidar does not ignore the pewm but only pwm=0 condition. To make it
stop you have to cut the power off.