ldrobotSensorTeam/ldlidar_stl_ros2 #12: How to solve it ? [ldlidar_stl_ros2_node-1] [ERROR] [1698204279.082635882] [STL27L]: ldlidar communication is abnormal.

wuh15, 2023-10-25:
ros2 launch ldlidar_stl_ros2 stl27l.launch.py 
[INFO] [launch]: All log files can be found below /home/xiaoyu/.ros/log/2023-10-25-11-24-36-000859-patrick-Precision-3460-2064107
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [ldlidar_stl_ros2_node-1]: process started with pid [2064108]
[INFO] [static_transform_publisher-2]: process started with pid [2064110]
[static_transform_publisher-2] [WARN] [1698204276.075204599] []: Old-style arguments are deprecated; see --help for new-style arguments
[ldlidar_stl_ros2_node-1] [INFO] [1698204276.081212779] [STL27L]: LDLiDAR SDK Pack Version is: v3.0.3
[ldlidar_stl_ros2_node-1] [INFO] [1698204276.081251821] [STL27L]: <product_name>: LDLiDAR_STL27L
[ldlidar_stl_ros2_node-1] [INFO] [1698204276.081256446] [STL27L]: <topic_name>: scan
[ldlidar_stl_ros2_node-1] [INFO] [1698204276.081259139] [STL27L]: <frame_id>: base_laser
[ldlidar_stl_ros2_node-1] [INFO] [1698204276.081261491] [STL27L]: <port_name>: /dev/ttyUSB0
[ldlidar_stl_ros2_node-1] [INFO] [1698204276.081263903] [STL27L]: <port_baudrate>: 921600
[ldlidar_stl_ros2_node-1] [INFO] [1698204276.081266528] [STL27L]: <laser_scan_dir>: Clockwise
[ldlidar_stl_ros2_node-1] [INFO] [1698204276.081269025] [STL27L]: <enable_angle_crop_func>: false
[ldlidar_stl_ros2_node-1] [INFO] [1698204276.081272418] [STL27L]: <angle_crop_min>: 0.000000
[ldlidar_stl_ros2_node-1] [INFO] [1698204276.081277139] [STL27L]: <angle_crop_max>: 0.000000
[ldlidar_stl_ros2_node-1] [INFO] [1698204276.081681053] [STL27L]: ldlidar node start is success
[static_transform_publisher-2] [INFO] [1698204276.082988739] [base_link_to_base_laser_stl27l]: Spinning until stopped - publishing transform
[static_transform_publisher-2] translation: ('0.000000', '0.000000', '0.180000')
[static_transform_publisher-2] rotation: ('0.000000', '0.000000', '0.000000', '1.000000')
[static_transform_publisher-2] from 'base_link' to 'base_laser'
[ldlidar_stl_ros2_node-1] [ERROR] [1698204279.082635882] [STL27L]: ldlidar communication is abnormal.
[ldlidar_stl_ros2_node-1] [LDS][INFO][1698204276.81653920][Actual BaudRate reported:921600]
[ERROR] [ldlidar_stl_ros2_node-1]: process has died [pid 2064108, exit code 1, cmd '/home/xiaoyu/workspace/ldlidar_ros2_ws/install/ldlidar_stl_ros2/lib/ldlidar_stl_ros2/ldlidar_stl_ros2_node --ros-args -r __node:=STL27L --params-file /tmp/launch_params_w2dldkmu --params-file /tmp/launch_params_1tk6ghqz --params-file /tmp/launch_params_vnkcsfmv --params-file /tmp/launch_params_8pmzdaea --params-file /tmp/launch_params__l1tj9su --params-file /tmp/launch_params_hp4ukfam --params-file /tmp/launch_params_9sa_mjlg --params-file /tmp/launch_params_3dojh6ll --params-file /tmp/launch_params__xdmxevb'].

arashrahmani, 2024-01-08:
I have the same issue

bdigney89, 2024-01-26:
i also have the same issue , using ros2 humble.

bperseghetti, 2024-02-06:
Are you sure it is connected to `/dev/ttyUSB0` could be on just about anything, IE ours is on: https://github.com/CogniPilot/b3rb_robot/blob/6457eb6ceed6ce7a54bc4bbcc3481a5af71802d3/b3rb_bringup/launch/laser.launch.py#L36

namato18, 2024-02-12:
I'm also having this issue, and yes it is indeed connected to /dev/ttyUSB0

bperseghetti, 2024-03-07:
> I'm also having this issue, and yes it is indeed connected to /dev/ttyUSB0
> 
> EDIT: I'm using a LD20 lidar, using the drivers for LD19 seemed to do the trick, used the drivers from this repo: https://github.com/AlgoPathfinder/sdk_ldrobotsensorteam_stl/tree/main

Yeah, if you (@wuh15 @arashrahmani @bdigney89) don't have an STL27L you should not run the STL27L launch file. I don't know how many people have that exact model but I doubt it's many? As it's not easy to source?

annpramodec, 2024-04-09:
Hello, I have a LD20 Lidar ... the wiki page links to the LD14 git repo, however when I run the launch file "ldlidar_ros_ws/src/ldldiar_sl_ros/launch/ld14.launch" , I am getting the error "ldlidar communication is abnormal", How do I solve this?

naidol, 2024-06-15:
Thanks gents.  Yes that LD20 driver does not work on ROS2(Humble).  Thanks to pointing me to the LD19 driver.  It works.
The driver can also be found on the suppliers website:
[YouYeeToo.com](https://wiki.youyeetoo.com/)

Scroll down to LIDAR and select LD19

https://drive.google.com/open?id=10nVCBdo7-LR3Cu8VlIEpKAykEDw_dEjG&usp=drive_fs

Xspencer6, 2024-12-01:
I am having the same issues, I am using LD06 on my Ubuntu 24.04 LTS OS Raspberry Pi 4B. I have ROS2 Jazzy installed, and 100% sure my lidar is connected to /dev/ttyUSB0. Tried launching ld06,ld19 and everything, same error. I don't know what to do now.

hippo5329, 2024-12-01:
Please run viewer_ld06 launch to check the driver.

https://github.com/hippo5329/ldlidar_stl_ros2

harmindersinghnijjar, 2025-01-24:
I would verify device connections using `ls /dev/tty*` for correct port names. I had to use a USB hub so that `ls /dev/tty*` outputted /dev/ttyUSB0 in the terminal as is listed in the launch file by default for communication to become normal.

Resources:

https://harminder.dev/blog/2025/01/23/setting-up-and-running-d500-lidar-kits-stl-19p-on-ros-2-jazzy/#1-communication-abnormal-error

https://harminder.dev/blog/2025/01/22/setting-up-and-running-a-roboclaw-based-ros-node-with-obstacle-avoidance/#troubleshooting

Plika-f, 2025-02-25:
This error can arise when you miss-connect your Tx and Rx cables. The tx from your lidar has to go into the Rx of the serial converter USB, and there's a chance you've connected the Tx to Tx of USB converter. And so, the pc cannot read it because there's nothing coming to it. Also, a reminder that the fourth cable of stl-27 is pwm cable of motor and not an Rx cable. But that shouldn't change much than to avoiding connecting it to the usb serial converter (my stl didn't come with a usb attached, so, i had to get it myself. You just ground the fourth cable, and not connect them all to the USB). 
Anyways, to make sure you're receiving data, you can connect your stl-27 and then run this code below: (Make sure you've arleady given the port access. And change the /dev/ttyTHS1 to your respective USB port. I'm using uart. This will work with stl-27L, i'm not sure what are the port settings of the other devices. But you could still try it out. If you see readable data, that means that you're receiving uart signals properly. If not, you should check your wiring.
What this code does is that it reads your serial port, and looks for the data encoding as provided by the development manual of the ldrobot stl-27L. I started this before i realized they have their own ROS manual. 

`import serial

# Open the UART port (change the port as needed, e.g., /dev/ttyTHS1)
serial_port = serial.Serial(
    port='/dev/ttyTHS1',  # Replace with your correct UART port
    baudrate=921600,
    bytesize=serial.EIGHTBITS,
    parity=serial.PARITY_NONE,
    stopbits=serial.STOPBITS_ONE,
    timeout=1  # Set a timeout to avoid blocking indefinitely
)

def read_uart_data():
    while True:
        byte = serial_port.read(1)  # Read one byte at a time
        if byte == b'\x54':  # Check for header (0x54 in hex)
            packet = byte + serial_port.read(31)  # Read the rest of the packet (assuming 32 bytes total)
            print("Raw Packet:", packet.hex())  # Print raw hex data
            
            # Extract key fields (modify based on exact format)
            version_length = packet[1]
            speed = int.from_bytes(packet[2:4], 'little')  # LSB | MSB
            start_angle = int.from_bytes(packet[4:6], 'little')  # LSB | MSB
            end_angle = int.from_bytes(packet[-4:-2], 'little')  # LSB | MSB
            timestamp = int.from_bytes(packet[-3:-1], 'little')  # Timestamp LSB | MSB
            crc = packet[-1]  # CRC byte

            print(f"Speed: {speed}, Start Angle: {start_angle/0.01}, End Angle: {end_angle/0.01}, Timestamp: {timestamp}, CRC: {crc}")

# Start reading data
read_uart_data()`