ldrobotSensorTeam/ldlidar_stl_ros2 #18: [async_slam_toolbox_node-1] laserrangescan contains 456 range readings, expected 451

0jeetparmar0, 2024-06-17:
[async_slam_toolbox_node-1] laserrangescan contains 456 range readings, expected 451

getting this error in slam_toolbox can someone help me out?

thanks and regards

booth-algo, 2024-06-18:
Hi @0jeetparmar0 , I am a LD19 user also encountering this issue. 

You can refer to issues #4 and #11 of this repo. There also seems to be an opened pull request that fixes this issue, but have not been merged by the (rather inactive) owners. 

Discussions about the root cause of this issue can also be found on slam_toolbox's repo (an issue raised back in 2020) and also similar [repos](https://github.com/Myzhar/ldrobot-lidar-ros2/issues/16) utilising LDROBOT LiDARs

Oyefusi-Samuel, 2025-05-09:
[INFO] [launch]: All log files can be found below /home/ubuntu/.ros/log/2025-04-25-16-40-50-113533-ubuntu-1992
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [async_slam_toolbox_node-1]: process started with pid [2014]
[async_slam_toolbox_node-1] [INFO] [1745599250.885564975] [slam_toolbox]: Node using stack size 40000000
[async_slam_toolbox_node-1] [INFO] [1745599251.640286308] [slam_toolbox]: Using solver plugin solver_plugins::CeresSolver
[async_slam_toolbox_node-1] [INFO] [1745599251.643999308] [slam_toolbox]: CeresSolver: Using SCHUR_JACOBI preconditioner.
[async_slam_toolbox_node-1] [WARN] [1745599251.704956234] [slam_toolbox]: minimum laser range setting (0.0 m) exceeds the capabilities of the used Lidar (0.0 m)
[async_slam_toolbox_node-1] Registering sensor: [Custom Described Lidar]
[async_slam_toolbox_node-1] LaserRangeScan contains 355 range readings, expected 354
[async_slam_toolbox_node-1] LaserRangeScan contains 355 range readings, expected 354
[async_slam_toolbox_node-1] LaserRangeScan contains 353 range readings, expected 354
[async_slam_toolbox_node-1] LaserRangeScan contains 355 range r