ldrobotSensorTeam/ldlidar_stl_ros2 #26: Lidar Communication is Abnormal

tracelarue, 2025-01-23:
Hello,

I recently bought the FHLD19 Plus from Youyeetoo on amazon, and when it arrived it has the STL-27L sticker label on it from LDRobot so I have been trying to set it up as such. I have followed exactly the instructions outlined in the README doc for ROS2 Humble and run into the following issue "lidar communication is abnormal". I have verified that data is transfering with minicom and made sure the proper port and baud were set in the parameters. Any help is appreciated! It may be worth nothing that their windows application also did not display any data for me

ERROR CODE BELOW :

trace@VirtualMachine3:~/lidar_ros2_ws$ ros2 launch ldlidar viewer_stl27l.launch.py
[INFO] [launch]: All log files can be found below /home/trace/.ros/log/2025-01-22-18-35-15-314576-VirtualMachine3-5950
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [ldlidar-1]: process started with pid [5951]
[INFO] [static_transform_publisher-2]: process started with pid [5953]
[INFO] [rviz2-3]: process started with pid [5955]
[static_transform_publisher-2] [WARN] [1737592515.421684072] []: Old-style arguments are deprecated; see --help for new-style arguments
[ldlidar-1] [INFO] [1737592515.438288937] [STL27L]: LDLiDAR SDK Pack Version is: v3.0.5
[ldlidar-1] [INFO] [1737592515.438434188] [STL27L]: <product_name>: LDLiDAR_STL27L
[ldlidar-1] [INFO] [1737592515.438449869] [STL27L]: <topic_name>: scan
[ldlidar-1] [INFO] [1737592515.438460547] [STL27L]: <frame_id>: base_laser
[ldlidar-1] [INFO] [1737592515.438470457] [STL27L]: <enable_serial_or_network_communication>: Enable serial
[ldlidar-1] [INFO] [1737592515.438480822] [STL27L]: <port_name>: /dev/ttyUSB0
[ldlidar-1] [INFO] [1737592515.438490996] [STL27L]: <port_baudrate>: 921600
[ldlidar-1] [INFO] [1737592515.438501064] [STL27L]: <server_ip>: 192.168.1.200
[ldlidar-1] [INFO] [1737592515.438511589] [STL27L]: <server_port>: 2000
[ldlidar-1] [INFO] [1737592515.438521470] [STL27L]: <laser_scan_dir>: Counterclockwise
[ldlidar-1] [INFO] [1737592515.438531744] [STL27L]: <enable_angle_crop_func>: false
[ldlidar-1] [INFO] [1737592515.438545209] [STL27L]: <angle_crop_min>: 135.000000
[ldlidar-1] [INFO] [1737592515.438562508] [STL27L]: <angle_crop_max>: 225.000000
[ldlidar-1] [INFO] [1737592515.438573910] [STL27L]: <measure_point_freq>: 21600
[static_transform_publisher-2] [INFO] [1737592515.447722976] [base_link_to_base_laser_stl27l]: Spinning until stopped - publishing transform
[static_transform_publisher-2] translation: ('0.000000', '0.000000', '0.180000')
[static_transform_publisher-2] rotation: ('0.000000', '0.000000', '0.000000', '1.000000')
[static_transform_publisher-2] from 'base_link' to 'base_laser'
[ldlidar-1] [INFO] [1737592515.482229151] [STL27L]: ldlidar node start is success
[rviz2-3] [INFO] [1737592515.832420247] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-3] [INFO] [1737592515.832506634] [rviz2]: OpenGl version: 4.5 (GLSL 4.5)
[rviz2-3] [INFO] [1737592515.928479459] [rviz2]: Stereo is NOT SUPPORTED
[ldlidar-1] [ERROR] [1737592515.983071792] [STL27L]: ldlidar communication is abnormal.
[ldlidar-1] [LDS][INFO][1737592515.482098688][Actual BaudRate reported:921600]
[ERROR] [ldlidar-1]: process has died [pid 5951, exit code 1, cmd '/home/trace/lidar_ros2_ws/install/ldlidar/lib/ldlidar/ldlidar --ros-args -r __node:=STL27L --params-file /tmp/launch_params_0nr2qla3 --params-file /tmp/launch_params_0gnmik3r --params-file /tmp/launch_params_rngyd7l8 --params-file /tmp/launch_params_7hms0fmv --params-file /tmp/launch_params_2wasfqna --params-file /tmp/launch_params_liyqkacs --params-file /tmp/launch_params_cu28jf7e --params-file /tmp/launch_params_yj2bcjjq --params-file /tmp/launch_params__d3r6bvf --params-file /tmp/launch_params_vvmivnoa --params-file /tmp/launch_params_lof1lrfg --params-file /tmp/launch_params_9pqti3rh --params-file /tmp/launch_params_za8htjwb'].
[INFO] [rviz2-3]: process has finished cleanly [pid 5955]
^C[WARNING] [launch]: user interrupted with ctrl-c (SIGINT)
[static_transform_publisher-2] [INFO] [1737592524.747662833] [rclcpp]: signal_handler(signum=2)
[INFO] [static_transform_publisher-2]: process has finished cleanly [pid 5953]

hippo5329, 2025-01-23:
Baudrate should be 21600.

hippo5329, 2025-01-23:
I guess it is still an LD19 with improved range. Please try

'''
ros2 launch ldlidar_stl_ros2 ld19.launch.py
'''

And check /scan topic and hz.

harmindersinghnijjar, 2025-01-24:
I would verify device connections using `ls /dev/tty*` for correct port names. I had to use a USB hub so that `ls /dev/tty*` outputted /dev/ttyUSB0 in the terminal as is listed in the launch file by default for communication to become normal.

Resources:

https://harminder.dev/blog/2025/01/23/setting-up-and-running-d500-lidar-kits-stl-19p-on-ros-2-jazzy/#1-communication-abnormal-error

https://harminder.dev/blog/2025/01/22/setting-up-and-running-a-roboclaw-based-ros-node-with-obstacle-avoidance/#troubleshooting

tracelarue, 2025-01-24:
@hippo5329 @harmindersinghnijjar 

Thanks for your help. I tried also the LD19 launch files but got the same error: LiDAR communication abnormal. I have also verified that the port is indeed /dev/ttyUSB0 at 921600 baud. I am using a virtual machine on VirtualBox and the usb passthrough is enabled and recognizing the device. Minicom shows that data is coming through the port also. I am also using the included USB to UART adapter with the driver installed both on the host machine and the virtual machine.

One thing I have noticed on the blue ring that spins on top of device, one of the segments in the circle is not lit. I am wondering if I have a defective unit.

hippo5329, 2025-01-24:
I have never used virtualbox for this, so I don't know if it will work. If your PC is windows, I would suggest docker or install native ubuntu on another partition or external drive.

hippo5329, 2025-01-24:
The log messages on their web page shown "ld06". You may try LD06.

hippo5329, 2025-01-24:
Native ubuntu is recommended when you need to interface to hardware, eg lidar.

tracelarue, 2025-01-25:
@harmindersinghnijjar 

Thank you for your response. I went through your link and instructions, but received the same errors.

I am absolutely positive that the device is connected to /dev/ttyUSB0 and that it is correct in the parameters as well as the baud rate. 

I have gone ahead and ordered a new one because I have tried every repository that exists for this thing as well as LD19 and whatever other variations are included and none have worked. Hopefully the new one will work.

hippo5329, 2025-01-27:
Try native ubuntu.

tracelarue, 2025-01-27:
I purchased an STL-19P from another seller which worked fine with the windows application from the LDRobot team. It must be something specifically about the STL-27l (or what youyeetoo advertises as a LD19 Plus). It is odd that the STL-27L which sells for $160 on DF robot is sold for $85 by youyeetoo. 🤔